Mechanical-Advantage/AdvantageScope #178: NT data text getting stacked on Ubuntu linux

DanPeled, 2024-09-08:
**Describe the bug**
I noticed this happens to me on my ubuntu laptop, but not on the team's windows DS,
on the NT data table preview on the side, all of the topic names are getting stacked on top of each other and makes it unreadable

**To Reproduce**
Steps to reproduce the behavior:

1. Open Advantage Scope through the `Start Tool` menu on WPILib vscode
2. Connect to the robot / sim / log file
3. View the side preview of the NT table

**Expected behavior**
All of the topic names are getting stacked on top of each other.

**Screenshots**
[image]
[image]

**Version (required):**
Version: 3.2.1
Distributor: WPILib
Platform: linux-x64
Build Date: 3/13/2024, 22:19:40 UTC
Electron: 26.2.1
Chromium: 116.0.5845.188
Node: 18.16.1

jwbonner, 2024-09-08:
This looks like the same problem as #99, which was the result of Intel driver issues. We can potentially switch Electron/Chromium versions if there have been related changes, but it seems like this is essentially a Chromium/Intel bug.

DanPeled, 2024-09-08:
> This looks like the same problem as #99, which was the result of Intel driver issues. We can potentially switch Electron/Chromium versions if there have been related changes, but it seems like this is essentially a Chromium/Intel bug.

ye seems to be like that

jwbonner, 2024-12-26:
Looks like this is a driver issue that can't be easily fixed on the AdvantageScope side. Closing the issue for now.

closed 2024-12-26 by jwbonner