Mechanical-Advantage/AdvantageScope #186: Add support for Chromebooks (AdvantageScope Lite in the browser)

TheTripleV, 2024-09-14:
For the past 6 years, every single high schooler I have mentored has had a chromebook that I've unfortunately had to treat as e-waste the moment they set foot into shop. It would be nice to have a version of AdvantageScope be supported purely in the browser that can be statically hosted on the rio (or 2027 mrc). I think this usecase will grow with ftc coming in in 2027.

I'm thinking the very basic lite features are

#### Yes
- Open file from disk (browser file api instead of fs)
- NT4 only connection
- Single window only

#### Maybe
- Run a rest api file server on the rio pointing at the log folder and let the browser query it

#### No
- No USB
- No NT3
- No support for youtube videos
- No sim support
- No export
- No publishing
- No layouts

blaze-developer, 2024-10-18:
This would be very impactful. If this was a feature, students could use Github Codespaces in Conjunction with this to run log replay, and view their replayed outputs in AdvantageScope using a simple Chromebook.

blaze-developer, 2025-09-09:
Isnt this completed?

jwbonner, 2025-09-09:
I think so, thanks for catching that.

closed 2025-09-09 by jwbonner