Mechanical-Advantage/AdvantageScope #188: Structs with schemas including types such as int, byte, and long don't deserialize correctly.

blaze-developer, 2024-09-16:
When connecting to a robot that publishes structured data types with WPILIB structs, they are only collapsable/you can see individual fields if they do not include any datatypes that are not doubles or floats.

Steps to reproduce the behavior:

1. Create a struct implementation for a data type that includes different types, like ints and bytes.
2. Log them through NT4 or AdvantageKit
3. Notice that you can not see any of the fields of the  serialized data if it includes any whole number types, (int, byte, long, etc)

I expected it to deserialize and show serialized ints and bytes to log data types with whole numbers as fields, however, it shows nothing instead.

jwbonner, 2024-09-16:
What are the type names you're using? Note that `int`, `byte`, and `long` are not valid types under the [struct spec](https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/doc/struct.adoc#standard-declaration). You probably want to use something like `char`, `int8`, `int32`, or `int64`.

blaze-developer, 2024-09-16:
Oop- yeah I got it completely wrong. I will try those! I apologize for me being dumb lol

closed 2024-09-17 by blaze-developer

blaze-developer, 2025-04-09:
I think you may have referenced the wrong issue but oh well

jwbonner, 2025-04-09:
Yeah, I referenced the issue using the ID in the other repo then transferred it after.