Mechanical-Advantage/AdvantageScope #197: Beta: Maximizing, then unmaximizing the window doesn't revert the window back to its unmaximized state

Gold856, 2024-09-26:
**Describe the bug**
When AdvantageScope is first started, it has a translucent background and rounded corners. These disappear when the window is maximized, but when the window is unmaximized, the window does not regain the translucent background and rounded corners.

**To Reproduce**
Steps to reproduce the behavior:

1. Open AdvantageScope.
2. Maximize the window.
3. Unmaximize the window.
4. Notice the lack of rounded corners and translucent background.

**Expected behavior**
I expected the rounded corners and translucency to return.

**Screenshots**
[image]

**Version (required):**
Version: 4.0.0-beta-1
Distributor: WPILib
Platform: win32-x64
Build Date: 9/26/2024, 24:43:48 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2024-09-26:
It looks like both this issue and #198 are related to https://github.com/electron/electron/issues/41824. I'll need to do some experimentation to see if there's a way to work around the issue.

jwbonner, 2024-10-05:
I've tried a couple of approaches, but unfortunately I can't find an effective workaround for that Electron bug. The only options would be to disable maximization (a regression from the 2024 version) or disable the new background material. I've gone with the latter option, and I'll monitor the Electron issue to see if we can eventually bring back the improved design.

The new title bar and menu bar are still enabled, since they don't appear to be causing any issues on their own.

closed 2024-10-05 by jwbonner