Mechanical-Advantage/AdvantageScope #201: Beta: The loading pulsating effect is incredibly distracting

person4268, 2024-09-26:
**Is your feature request related to a problem? Please describe.**
I've just tried the beta, and while I like it overall, the pulsating effect on the left when keys are loading is distracting me pretty badly, especially when I've already grabbed the things I need and I'm trying to focus on the main view on the right. Additionally, each individual instance of the effect desyncs, making the effect even worse. This isn't the most significant issue, but it is genuinely throwing me off a fair bit.

**Describe the solution you'd like**
Replacing the effect with something else. I'm not sure exactly what, but preferably something that isn't animated and is fairly discreet. 

**Describe alternatives you've considered**
Slowing down and/or making sure the pulsating in sync would help, but I just find the effect intrinsically distracting when I'm trying to find a key in the tree or whatnot.

**Additional context**

(ignore me realizing everything had already loaded at the beginning)
https://github.com/user-attachments/assets/a8557844-2ae0-4588-a62e-b311d0faa40e

jwbonner, 2024-09-26:
The main requirement is that it needs to be very obvious to the user that something is happening, since otherwise it looks like the data isn't available. Below is a slightly more subtle option that uses the cursor and an opacity shift to indicate that progress is being made.

If this looks better to you, we can try it in the next beta and get feedback from more users.

https://github.com/user-attachments/assets/d8babaaf-b014-4c0c-837c-361ed211525b

person4268, 2024-09-26:
Yeah, that's a lot better and more intuitive imo. Signals pretty clearly that "these aren't usable" (yet) while I couldn't get an instant read on what the original effect was trying to indicate for a minute. Thanks for fixing that. My other qualms are very minor and almost nitpicky and fairly subjective in comparison, but I can write them out if you want them.

jwbonner, 2024-09-26:
I’d welcome hearing any feedback you have, and we can figure things out if some are subjective.

jwbonner, 2024-10-05:
Feel free to open a new issue if you have additional feedback.

PR #213 (2024-10-05): Add alternative loading indicator for fields (#201)
Closes #201
closed 2024-10-05 by jwbonner