Mechanical-Advantage/AdvantageScope #202: Auto Downloading logs toggle

shoryamalani, 2024-09-27:
Hey, our team and a few others that I know about almost always forget to download logs at the end of the meeting/ during events but still need them

A toggle that just has the ability to "download new logs" which just syncs to a folder with the last day or two's logs auto synced would be amazing. having the ability to not continuously download all logs would be especially good because not everyone has space on their computer.

jwbonner, 2024-09-27:
Seems like a duplicate of #153?

Regardless, my primary concern with a feature like this is bandwidth usage. Keep in mind that the "logging" NT mode is disabled by default because sending all of the robot's logged data over the network in real time requires a significant amount of bandwidth (enough that it can interfere with other tasks). Automatically downloading logs would average out to a similar amount of bandwidth usage, which is especially concerning if you had multiple clients. I would say the minimum is that this background task shouldn't run when the robot is enabled, probably with a bandwidth limit in other cases. I think there would be a way to make this work with a bit of thought though.

Also, a few other options that are worth considering;
- If the primary concern is just forgetting to download logs, an alternative would be having a button in the title bar that appears when new logs are available (and is one click to download them).
- Given that the Driver Station already supports downloading logs, is this a more appropriate feature to add there? The DS seems better suited for background tasks since it's left running at all times (while AdvantageScope is closed during matches), and users expect that the DS will use a significant amount of resources when connected to the robot.

Just my initial thoughts, I haven't had time to think about this in more depth yet.

shoryamalani, 2024-09-27:
"If the primary concern is just forgetting to download logs, an alternative would be having a button in the title bar that appears when new logs are available (and is one click to download them)."
- I think thats a great approach to the problem
Also, a lot of people run Ascope on a programmers laptop that is separate from a dedicated driver station which never goes home with programmers

blaze-developer, 2024-10-12:
If a feature was implemented, maybe allow for filters for logs, for example only logs that are actual games and not just software pit-checks.

sebasptsch, 2024-10-14:
I think @shoryamalani 's approach explains the utility well. Typically our driver station laptops aren't connected to the internet often (as they're often connected to the robot wirelessly). Having advantagescope download the files and then connecting to the internet (after downloading changed logs) and using a 3rd party sync utility would be great.