Mechanical-Advantage/AdvantageScope #203: Forever Exporting

Eaglegrant, 2024-09-27:
**Describe the bug**
After exporting one log, it seems to never finish that export (even when it has!). The log can be opened on other computers, and viewed in full, showing no lost/unexported data. However, AdvantageScope believes it hasn't finished, and doesnt ever end. I can't export more logs without fully shutting down AScope and restarting it.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect to Robot
2. Click on 'Export Data'
3. Hit ok.
4. Wait for approximately 1-3 minutes.
5. Clock on 'Export Data'
6. See error. (This is incorrect. The export is complete, the log is 100% full with every sample.)

**Expected behavior**
I should be able to export another log!

**Version (required):**
Version: 4.0.0-beta-1
Distributor: FRC 6328
Platform: win32-x64
Build Date: 9/26/2024, 24:41:47 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2024-10-05:
I haven't been able to recreate this in my testing. Can you upload a copy of the log you're exporting, along with a screenshot of the export settings?

jwbonner, 2024-12-26:
We haven't been able to recreate this issue, so I'm closing the issue for now. Feel free to reopen if you have a log file and export settings where you can consistently recreate the bug.

closed 2024-12-26 by jwbonner