Mechanical-Advantage/AdvantageScope #204: High GPU utilization at idle

rzblue, 2024-09-28:
AdvantageScope 2025 has significant continuous GPU utilization on Windows even when not displaying anything and/or in the background. Screenshot shows 2 AdvantageScope instances running; The top is 2025, the bottom is 2024 (both wpilib packaged versions). Both instances in the screenshot are open side-by-side in the foreground (utilization does not change when in background/not visible).

[image]

jwbonner, 2024-10-05:
There were several tabs (notably the line graph) which supported on-demand rendering in 2024 but I neglected to reimplement when rewriting them. I pushed a fix.

closed 2024-10-05 by jwbonner

rzblue, 2024-10-11:
Sweet, that's much better!