Mechanical-Advantage/AdvantageScope #205: App Crashes when Scrolling on Windows Explorer

DelgLe, 2024-09-28:
**Describe the bug**
A clear and concise description of what the bug is.

I just downloaded the beta. I went to try to open a file I had downloaded locally from my one drive. As soon as I opened the folder containing the log files and hoot files that I wanted to open it crashed, and from then, after navigating to that folder, it automatically crashed on opening file explorer from 'Open Logs'
 
**To Reproduce**
Steps to reproduce the behavior:

1. Start App
2. Click file
3. Click Open Logs
4. As soon as file explorer is opened Advantage Scope crashes

**Expected behavior**
I would be able to open a log file from a folder.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Version (required):**
Copy the contents of the "About AdvantageScope" window and paste it here. On Windows/Linux, click "Help" > "About AdvantageScope". On macOS, click "AdvantageScope" > "About AdvantageScope" in the menu bar.

Version: 4.0.0-beta-1
Distributor: FRC 6328
Platform: win32-x64
Build Date: 9/26/2024, 24:41:47 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
Add any other context about the problem here.

I'm using my work laptop at the moment. 
'Add new logs' has the same behavior.

jwbonner, 2024-10-05:
In order to debug this, it would be helpful to see the console output from AdvantageScope. You'll need to run it from the command line. If it's installed in the default user location, you should be able to launch it by running:

```
%userprofile%\AppData\Local\Programs\advantagescope\AdvantageScope.exe
```

After attempting to open a file, there will hopefully be a message printed to the console with more details.

jwbonner, 2024-12-26:
We haven't been able to recreate this, so I'm closing the issue for now. Feel free to reopen if you have a console log or can consistently recreate it on other devices.

closed 2024-12-26 by jwbonner