Mechanical-Advantage/AdvantageScope #208: Integrated data: Add view of current numeric total

jamesdooley4, 2024-09-30:
**Is your feature request related to a problem? Please describe.**
When using the new Integrated view in Line Graph, it's not possible to see the current integrated total exact value.

**Describe the solution you'd like**
Add the current integrated total value as a number in the Left or Right axis details view, either replace or next to the existing view of the current measured value (10) in the picture here:
[image]

**Describe alternatives you've considered**
Currently I eyeball the value using the graph view, but that often isn't very precise

**Additional context**
Nothing else

jwbonner, 2024-10-05:
The value displayed under the axis is the integrated value at whatever time is selected or hovered on the graph. Is there something else you're looking for?

jamesdooley4, 2024-10-05:
When opening a new file today, it does seem to be worked as described. I'll play with it to see what I was doing to get it into the broken state.

closed 2024-12-26 by jwbonner