Mechanical-Advantage/AdvantageScope #209: Beta: Problems with loading models, and 3D field

blaze-developer, 2024-09-30:
When loading a model for a 3D Field visualization, for example a robot model, it loads the model after a few seconds.

However, before the beta, models seemed to cache or save for every app session, resulting in low loading times after the initial load. Now, every time I change sources, swap the models back and forth, and even just minimize or maximize the window, the models go through a complete reload, taking multiple seconds that make it really inconvenient for watching live sources.

Also, Cinematic mode is extremely laggy at least on my PC, but it ran a lot faster in the previous version.

As well as this, many textures on models dont seem to load, loading as just white. This is dismissable, but something I noticed.

[image]

jwbonner, 2024-10-06:
**TLDR; I pushed some changes to address these issues. I'll release those in the next beta, and feel free to reopen the issue if you're still having trouble at that point.**

The way that field and robot models are loaded in 2025 is different than 2024. CAD models often contain an enormous number of small meshes, which require separate draw calls. This meant that performance on the 2024 version was almost always CPU limited.

For the 2025 version, each model is now merged into (effectively) a single mesh which can be sent to the GPU as a single draw call. This change significantly improves performance on many devices that were CPU limited on the previous version. The downside of this approach is that the individual components of each model can no longer be frustum culled (so the GPU always needs to render the full model). This works fine for robot models since you typically have the entire model on screen, but it's not ideal for the field model where you're typically looking at only part of it. Cinematic mode makes this even worse because there are additional cameras used for shadow calculations.

I just pushed a change which should address this issue by chunking the field model into smaller pieces, each of which can be culled when out of frame. This means that there are still significantly fewer draw calls than in 2024, but the GPU doesn't need to render the whole field all the time.

The increased loading time for models is a result of this mesh merging operation along with a simplification step that removes small objects from the models. This is a tradeoff between longer loading time and improved runtime performance. The biggest issue seems to be that models are often reloaded unnecessarily right now, such as when objects are temporarily hidden or rearranged in the list. I pushed a couple of changes that should reduce how often models need to be reloaded.

closed 2024-10-06 by jwbonner

blaze-developer, 2024-10-16:
Is it possible to turn off model simplification? I have multiple missing pieces in AdvantageScope with the new version, and it results in visibly floating parts. It would be nice to either have an option to disable this, or to set your own standard for how small objects have to be to be deleted.