Mechanical-Advantage/AdvantageScope #210: Component position offsets not affecting model

DanPeled, 2024-10-01:
**Describe the bug**
When changing the the position offset on a custom robot model's component, the position in the field of the component isnt being affected 

**Version (required):**
Version: 4.0.0-beta-1
Distributor: FRC 6328
Platform: linux-x64
Build Date: 9/26/2024, 24:41:47 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2024-10-05:
Are you following the setup process [here](https://docs.advantagescope.org/more-features/custom-assets#setup-process)? The `zeroedRotations` and `zeroedPosition` keys don't affect the component positions unless a component pose is published.

DanPeled, 2024-10-06:
Oh, I figured it might be something like that, thanks!

closed 2024-10-06 by jwbonner