Mechanical-Advantage/AdvantageScope #211: AdvantageScope 2025 Beta not opening

falOn-Dev, 2024-10-03:
After installing the 2025 AdvantageScope beta, once I closed it, it no longer would open, when I click on it, 4 new processes pop up in Task Manager, all labelled "AdvantageScope", but the app still does not open, the only thing that seemed to fix it was uninstalling and reinstalling, which caused it to open again, but once I closed it, it wouldn't open again.

Using the latest beta, on x86 64bit Windows 10

falOn-Dev, 2024-10-03:
The only way I've found to open it, is to open an admin PowerShell terminal, and run the exe from there

falOn-Dev, 2024-10-03:
That doesn't work 100% of the time unfortunately. the error I get is

```
[22900:1003/134236.681:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[22900:1003/134236.681:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[22900:1003/134236.681:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[22900:1003/134236.684:ERROR:disk_cache.cc(208)] Unable to create cache
[22900:1003/134236.684:ERROR:gpu_disk_cache.cc(713)] Gpu Cache Creation failed: -2
[22900:1003/134236.684:ERROR:disk_cache.cc(208)] Unable to create cache
[22900:1003/134236.684:ERROR:gpu_disk_cache.cc(713)] Gpu Cache Creation failed: -2
[22900:1003/134236.684:ERROR:disk_cache.cc(208)] Unable to create cache
[22900:1003/134236.684:ERROR:gpu_disk_cache.cc(713)] Gpu Cache Creation failed: -2
[6872:1003/134236.684:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[6872:1003/134236.684:ERROR:disk_cache.cc(208)] Unable to create cache
```

jwbonner, 2024-12-26:
Thus far we haven't been able to recreate this bug on other devices, so I'm closing the issue for now. Feel free to reopen if you're able to recreate it consistently on other devices.

closed 2024-12-26 by jwbonner

falOn-Dev, 2024-12-27:
No stress, I ended up switching to linux for unrelated reasons, but it works fine, I think my windows install was borked in some weird way :shrug:

jlange6648, 2025-03-04:
I continuously run into this issue on my PC. the only way to fix it seems to be a reboot, then it will launch. Until it starts acting up again

```
[29372:0304/181329.993:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[29372:0304/181329.993:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[29372:0304/181329.993:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[29372:0304/181330.006:ERROR:disk_cache.cc(208)] Unable to create cache
[29372:0304/181330.006:ERROR:gpu_disk_cache.cc(713)] Gpu Cache Creation failed: -2
[29372:0304/181330.007:ERROR:disk_cache.cc(208)] Unable to create cache
[29372:0304/181330.007:ERROR:gpu_disk_cache.cc(713)] Gpu Cache Creation failed: -2
[29372:0304/181330.008:ERROR:disk_cache.cc(208)] Unable to create cache
[29372:0304/181330.009:ERROR:gpu_disk_cache.cc(713)] Gpu Cache Creation failed: -2
[13776:0304/181330.051:ERROR:cache_util_win.cc(20)] Unable to move the cache: Access is denied. (0x5)
[13776:0304/181330.054:ERROR:disk_cache.cc(208)] Unable to create cache
```