Mechanical-Advantage/AdvantageScope #212: Records with the same timestamp are not displayed

Gold856, 2024-10-03:
**Describe the bug**
Unique records with the same timestamp are not shown in the console or table view.

**To Reproduce**
Steps to reproduce the behavior:

1. Open this log (file extension changed): 
[wpilog_b90345366c515336.log](https://github.com/user-attachments/files/17239339/wpilog_b90345366c515336.log)
2. Open a table/console view.
3. Place the test entry inside.
4. Observe that there are not 100 records.

**Expected behavior**
There should be 100 records, counting from 0 to 99.

**Screenshots**
[image]
Printlog showing more records and the same timestamp:
[image]

**Version (required):**
Version: 4.0.0-beta-1
Distributor: WPILib
Platform: win32-x64
Build Date: 9/26/2024, 24:43:48 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
Also reproduced on the 2024 release.

rzblue, 2024-10-03:
This may be an issue with the new console logging feature; data is read in bulk from the console output file and written to the datalog at most every 100ms, and each newline starts a new record, so there can be multiple updates in a row with the same timestamp.

rzblue, 2024-10-04:
This can be closed, we can change the console log behavior

closed 2024-10-04 by Gold856