Mechanical-Advantage/AdvantageScope #217: Video loading on linux

blaze-developer, 2024-10-07:
On both the 4.0.0 beta and 3.2.1, when clicking the youtube loading button, it just loads infinitely. Loading a local file works however, but near the end of video loading, it errors loading the video, after checking my filesystem, the seeming issue seems to be that my temporary folder is getting extremely full from just a single video, and then erroring while trying to write the video to the filesystem where it is impossible to write to.

Why does advantagescope need to save videos into temporary folders if its already loading from the disk? Temporary folders on many machines are limited in size, by default, mine is 80 megabytes.

[image]

blaze-developer, 2024-10-07:
YouTube video urls would still be a remaining issue, I have checked the console and network tab for when I click the paste button and no network requests start, and no console messages log

blaze-developer, 2024-10-11:
TBA got updated with match videos, so now after testing, the same problem exists for TBA video loading as well.

jwbonner, 2024-10-15:
You won't see any network requests or console messages when downloading videos because all of that processing is handled by the main process, not the renderers. However, YouTube and TBA downloads are likely to encounter the same filesystem issue.

jwbonner, 2024-10-15:
I'm attempting to recreate this by writing the video cache to a filesystem with limited free space, but ffmpeg appears to produce an error that is propagated properly to the rendered when no more space is available. Can you send the console output from the main process when loading a local video? You'll need to run AdvantageScope from the command line to see that output.

blaze-developer, 2024-10-15:
There appears to be two problems here, the filesystem issue, and an issue preventing online videos to load, and infinite loading with no visible progress. I will run both through the terminal, and send the results.

blaze-developer, 2024-10-15:
There appears to be two problems here, the filesystem issue, and an issue preventing online videos to load, and infinite loading with no visible progress. I will run both through the terminal, and send the results.

With a local video:
```
*** START FFMPEG OUTPUT ***
ffmpeg version 5.0.1-static https://johnvansickle.com/ffmpeg/  Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)
  configuration: --enable-gpl --enable-version3 --enable-static --disable-debug --disable-ffplay --disable-indev=sndio --disable-outdev=sndio --cc=gcc --enable-fontconfig --enable-frei0r --enable-gnutls --enable-gmp --enable-libgme --enable-gray --enable-libaom --enable-libfribidi --enable-libass --enable-libvmaf --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-librubberband --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libvorbis --enable-libopus --enable-libtheora --enable-libvidstab --enable-libvo-amrwbenc --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libdav1d --enable-libxvid --enable-libzvbi --enable-libzimg
  libavutil      57. 17.100 / 57. 17.100
  libavcodec     59. 18.100 / 59. 18.100
  libavformat    59. 16.100 / 59. 16.100
  libavdevice    59.  4.100 / 59.  4.100
  libavfilter     8. 24.100 /  8. 24.100
  libswscale      6.  4.100 /  6.  4.100
  libswresample   4.  3.100 /  4.  3.100
  libpostproc    56.  3.100 / 56.  3.100
Input #0, matroska,webm, from '/home/bryce/F2 2024 PNW Oregon Girl's Gen Event   WIDE [wJGrvpP1Vv4].mkv':
  Metadata:
    COMPATIBLE_BRANDS: iso6avc1mp41
    MAJOR_BRAND     : dash
    MINOR_VERSION   : 0
    ENCODER         : Lavf61.1.100
  Duration: 00:06:43.03, start: 0.000000, bitrate: 3659 kb/s
  Stream #0:0: Video: h264 (High), yuv420p(tv, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 60 fps, 60 tbr, 1k tbn (default)
    Metadata:
      HANDLER_NAME    : ISO Media file produced by Google Inc.
      VENDOR_ID       : [0][0][0][0]
      DURATION        : 00:06:43.000000000
  Stream #0:1(eng): Audio: opus, 48000 Hz, stereo, fltp (default)
    Metadata:
      DURATION        : 00:06:43.028000000
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> mjpeg (native))
Press [q] to stop, [?] for help
[swscaler @ 0x15b1b880] [swscaler @ 0x15b29a80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15b6ce80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15bafe00] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15bf2d80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15c33e00] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15c76d80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15cb7e00] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15cfad80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15d3dd00] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15d7ed80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15dc1d00] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15e04c80] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x15b1b880] [swscaler @ 0x15e45d00] deprecated pixel format used, make sure you did set range correctly
Output #0, image2, to '/tmp/advantagescope-videos/1ss4oh1ou00p0pe2mcse59l7bhs18hqp/%08d.jpg':
  Metadata:
    COMPATIBLE_BRANDS: iso6avc1mp41
    MAJOR_BRAND     : dash
    MINOR_VERSION   : 0
    encoder         : Lavf59.16.100
  Stream #0:0: Video: mjpeg, yuvj420p(pc, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 60 fps, 60 tbn (default)
    Metadata:
      HANDLER_NAME    : ISO Media file produced by Google Inc.
      VENDOR_ID       : [0][0][0][0]
      DURATION        : 00:06:43.000000000
      encoder         : Lavc59.18.100 mjpeg
    Side data:
      cpb: bitrate max/min/avg: 0/0/200000 buffer size: 0 vbv_delay: N/A
av_interleaved_write_frame(): No space left on deviceitrate=N/A speed= 3.7x
frame=11819 fps=223 q=2.0 Lsize=N/A time=00:03:16.81 bitrate=N/A speed=3.71x
video:4407766kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
Conversion failed!

*** END FFMPEG OUTPUT ***
[1180606:1014/214253.948506:ERROR:browser_main_loop.cc(283)] GLib-GObject: ../glib/gobject/gsignal.c:2685: instance '0x159406006680' has no handler with id '61813'
```

When clicking the TBA button or YT button:
nothing seems to be logged in the console, its as if the button just starts an animation and does nothing.

blaze-developer, 2024-10-15:
This behavior is odd to me, because the video loaded is under 200 megabytes large, and it somehow fills up my entire temporary folder which is configured to have 6 gigabytes of space.

jwbonner, 2024-10-15:
It looks like the error with the local file matches what I'm seeing, so I'll look at how I can work around that. It's expected that AdvantageScope will produce files much larger than the original video. It needs to decode the full video ahead of time to enable efficient scrubbing, since the type of back-and-forth video scrubbing required by AdvantageScope works very poorly when decoding videos in real time (most video formats are designed only for playback in one direction).

I'm not sure about the YouTube issue, since that match video downloads correctly for me. Have you tried running it with different videos, computers, or networks? It would be helpful to try to narrow down the conditions for recreating this.

blaze-developer, 2024-10-15:
I have tried running the TBA and Youtube loading with multiple different videos on this machine, and none have worked. However running on a windows software laptop from my team, it works as expected. I haven't tried other machines running Linux however, but on my desktop it does not work. I can try running on a different Linux machine but I would have to borrow some kind of device. Has anyone attempted to recreate this on a linux system?

jwbonner, 2024-10-16:
Videos from YouTube and TBA are loading correctly for me on Ubuntu 22.04. I've had issues in the past related to DNS resolution, but that always prints an error from ffmpeg (like you were seeing from a local video). What distribution and version of Linux are you running?

blaze-developer, 2024-10-16:
I am running Arch Linux with a rolling release with Kernel version 6.10.7

blaze-developer, 2024-10-16:
I have a seperate windows hard drive, so tonight I will test on the same machine with the windows operating system to see if it is machine specific or operating system specific.

blaze-developer, 2024-10-17:
Update, I have run this on windows on the same pc, and instead of it doing nothing, I get an error straight away.

```
ffmpeg version 5.0.1-essentials_build-www.gyan.dev Copyright (c) 2000-2022 the FFmpeg developers                                                                                                                                                                                 built with gcc 11.2.0 (Rev7, Built by MSYS2 project)                                                                                                                                                                                                                           configuration: --enable-gpl --enable-version3 --enable-static --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-lzma --enable-zlib --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-sdl2 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libaom --enable-libopenjpeg --enable-libvpx --enable-libass --enable-libfreetype --enable-libfribidi --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libmfx --enable-libgme --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libtheora --enable-libvo-amrwbenc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-librubberband                                                                                                                                                                          libavutil      57. 17.100 / 57. 17.100                                                                                                                                                                                                                                         libavcodec     59. 18.100 / 59. 18.100                                                                                                                                                                                                                                         libavformat    59. 16.100 / 59. 16.100                                                                                                                                                                                                                                         libavdevice    59.  4.100 / 59.  4.100                                                                                                                                                                                                                                         libavfilter     8. 24.100 /  8. 24.100                                                                                                                                                                                                                                         libswscale      6.  4.100 /  6.  4.100                                                                                                                                                                                                                                         libswresample   4.  3.100 /  4.  3.100                                                                                                                                                                                                                                         libpostproc    56.  3.100 / 56.  3.100                                                                                                                                                                                                                                       [https @ 000002959e25dfc0] HTTP error 403 Forbidden                                                                                                                                                                                                                            https://rr2---sn-nx57ynlk.googlevideo.com/videoplayback?expire=1729156713&ei=CYIQZ5yeLt3BsfIPoqWBwAQ&ip=2601%3A1c0%3A4600%3A61a0%3A64bf%3A31e8%3A888f%3Af655&id=o-AISuInXGt8OZxNgQsRHPMB8Hif9_0aA6mTQGwm6f7yD3&itag=299&aitags=134%2C136%2C160%2C243%2C298%2C299&source=youtube&requiressl=yes&xpc=EgVo2aDSNQ%3D%3D&met=1729135113%2C&mh=Zc&mm=31%2C29&mn=sn-nx57ynlk%2Csn-nx5s7n7y&ms=au%2Crdu&mv=m&mvi=2&pl=38&rms=au%2Cau&initcwndbps=1667500&bui=AXLXGFQmfB8fXlFhFPIYga9JpK8B7HRBmTv1zZ-Dp6lqPEPjTDUSF8VHZGN96V4mh34NAKKWr_7mlxrM&spc=54MbxXg5u0k3O6Q3yeFke0s4liVJrMdtetbKunNt0Ze1mSyOhEaS6QVfzw_l&vprv=1&svpuc=1&mime=video%2Fmp4&ns=N_ZbbXqNO68DFB4M7iBIJnIQ&rqh=1&gir=yes&clen=149343234&dur=323.000&lmt=1728151882785112&mt=1729134568&fvip=3&keepalive=yes&fexp=51264008%2C51300760%2C51312688&c=WEB&sefc=1&txp=6209224&n=GGdOdTOn9-fdxelG&sparams=expire%2Cei%2Cip%2Cid%2Caitags%2Csource%2Crequiressl%2Cxpc%2Cbui%2Cspc%2Cvprv%2Csvpuc%2Cmime%2Cns%2Crqh%2Cgir%2Cclen%2Cdur%2Clmt&lsparams=met%2Cmh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Crms%2Cinitcwndbps&lsig=ACJ0pHgwRgIhAOZYDQ3wSJNlMg3NdODSzABc5cbqyi6N2U-bMM-Ap7c0AiEAm3lhBX5_AruxN4C0nTEeBsF5QGji9-lRrC40wNJslaM%3D&sig=AJfQdSswRAIgGL5RlEkEJkHGQSdxFDoyGEYnvwvVvg51pg51RLrdgc4CICmK11FMXNTrPH6Pkjw-_K0WdL7i2nUb49hiD1DUcjVd: Server returned 403 Forbidden (access denied)
```

Im not sure what would cause this to happen though, and this error didnt show up on the linux OS, it just did nothing.

jwbonner, 2024-10-18:
I added logic to switch to a fallback directory on the main drive if the temporary volume is out of space, which should address the issue with local files.

My best guess is that the YouTube/TBA problems are related to some type of issue with the YouTube download API, which of course isn't officially supported by Google. There could be any number of underlying reasons, whether it's related to user agent, IP address, etc. It seems like it's still working in many cases, but ultimately YouTube is very unpredictable and I know they use tactics like checking the IP address to prevent some types of misuse. We will continue to update the relevant dependency ([ytdl-core](https://www.npmjs.com/package/@distube/ytdl-core)) as new versions are available, but otherwise I think this issue falls under the warning in the docs that "YouTube and TBA video download may failed unexpectedly due to changes on YouTube's servers."

jwbonner, 2024-12-26:
We haven't been able to recreate issues with YouTube downloading on other devices, so I'm closing this issue for now.

closed 2024-12-26 by jwbonner