Mechanical-Advantage/AdvantageScope #221: Google Drive or Dropbox sync support.

blaze-developer, 2024-10-11:
It would be WONDERFUL if we could have an option that puts log files onto a drive folder to be shared across a team. These could be organized by directories named with github branch names, for example, event_firstchance or just based on the time.  Right now its hard to distribute log files amongst multiple software students and mentors that all want access as soon as a match is over to debug and test.

You could possibly add a google drive or dropbox api key setting.

blaze-developer, 2024-10-11:
Well, at least for my team we only connect to the robot in the pits with USB or Ethernet, which still allows for internet access simultaneously. I see your point though. Maybe it could save logs into a folder and then either wait for an Internet connection, or manually sync it once you are connected

jwbonner, 2024-10-15:
After further discussion, we don't think that this needs to be a feature distinct from #202. Users would be free to use a cloud syncing tool of their choice to automatically upload log files downloaded by AdvantageScope, which will be both more reliable and more customizable than including the functionality in AdvantageScope.

closed 2024-10-15 by blaze-developer