Mechanical-Advantage/AdvantageScope #223: Values show up in the left pane but not inside the tabs

Emma03L, 2024-10-14:
**Describe the bug**
values exist and update in the left pane but when put into tabs no value is shown next to the name or in the graph.

**To Reproduce**
Steps to reproduce the behavior:
I am not sure how to reproduce it but closing and reopening the application seemed to solve it.

**Expected behavior**
I expected the values to show up as graphs correctly.

**Screenshots**
[image]
[image]
The value `AbsoluteEncoderAngle` is present and is updating on the left (under Hood) but isn't doing anything in the graphs.

**Version (required):**
Version: 4.0.0-beta-1
Distributor: FRC 6328
Platform: win32-x64
Build Date: 9/26/2024, 24:41:47 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2024-10-15:
It looks like the line graph isn't rendering at all. If this occurs again, it would be helpful if you can check for errors in the developer console (go to "View" > "Toggle Developer Tools").

Without being able to recreate this, there's very little I can do to fix it.

jwbonner, 2024-10-28:
My best guess is that this is a duplicate of #238, given that the screenshots show an image of the statistics tab. If the issue is not fixed on the next beta release of AdvantageScope, feel free to reopen.

closed 2024-10-28 by jwbonner