Mechanical-Advantage/AdvantageScope #230: RLOG Questions

blaze-developer, 2024-10-16:
I noticed that in your team's 2024 robot code, you use an RLOG server for publishing data. However beforehand I remember RLOG being an old format that shouldn't be used anymore. Are there any advantages to RLOG over NT?

Thanks :3

jwbonner, 2024-10-16:
RLOG files are not supported by AdvantageKit, but RLOG live streaming is supported as an alternative to NetworkTables. Our recommendation to all teams is to use NetworkTables for live streaming. RLOG streaming currently had stability issues and we do not expect to change our recommendation to teams in the foreseeable future, especially as NetworkTables continues to improve in future WPILib releases.

blaze-developer, 2024-10-16:
Interesting, was there a problem with NT that you were attempting to solve with RLOG streaming?

jwbonner, 2024-10-16:
NetworkTables 4 has had stability issues over the past few years in certain circumstances. Most of those have been ironed out by now, but we implemented an updated version of RLOG as a backup. But again, it will almost certainly be less stable than NetworkTables for most users and we don't plan on providing support for it.

closed 2024-10-16 by blaze-developer