Mechanical-Advantage/AdvantageScope #231: Add option to disable model simplification

jwbonner, 2024-10-16:
Is it possible to turn off model simplification? I have multiple missing pieces in AdvantageScope with the new version, and it results in visibly floating parts. It would be nice to either have an option to disable this, or to set your own standard for how small objects have to be to be deleted. I apologize if I should have started a new issue, but this seemed to be relevant to the previous conversation.

_Originally posted by @blaze-developer in https://github.com/Mechanical-Advantage/AdvantageScope/issues/209#issuecomment-2417889103_

blaze-developer, 2024-10-18:
What does the documentation mean by "mesh name" is this the name of the mesh in the glb file? Is there a simple way to add that to a glb file?

jwbonner, 2024-10-18:
You can edit GLB files (including mesh names) using CAD Assistant, or your CAD program may use existing object names that can be edited.

blaze-developer, 2024-10-18:
Ok thanks so much! :3

commit bd3dc7cb (2024-10-18): Add option to disable simplification of robot assets (closes #231)

Also document method of disabling simplification for individual meshes