Mechanical-Advantage/AdvantageScope #234: Custom Assets link near bottom of README still links to old documentation.

blaze-developer, 2024-10-18:
Scroll to the bottom of the readme under assets section.

jwbonner, 2024-10-18:
Thanks, I fixed the link in the README. The emoji on the line graph page looks right to me, it's a map icon that matches the odometry tab. Emoji rendering is dependent on operating system, so is it possible that your OS doesn't support that icon?

[image]

blaze-developer, 2024-10-18:
I meant the blue arrow emoji on that tab is a different emoji than the others. Perhaps my noto fonts make it more obvious than your fonts.

jwbonner, 2024-10-18:
I've confirmed that the right arrow uses exactly the same unicode symbol on several tabs. Can you send a screenshot of what you're seeing?

blaze-developer, 2024-10-18:
I meant the blue arrow emoji on that tab is a different emoji than the others. Perhaps my noto fonts make it more obvious than your fonts. It appears as a yellow boxed arrow and not a blue arrow on a chromeOS laptop and not any of my other devices.

blaze-developer, 2024-10-18:
[image]

blaze-developer, 2024-10-18:
This renders correctly on other tabs, it seems to be only that one.
[image]

jwbonner, 2024-10-18:
I don't think there's anything I can do about a browser that is failing to render the page as served. The styling of those pages is identical, and changing the styling for a single page is nontrivial even if I wanted to do so. I don't see any evidence of an issue with the page itself, so I'm going to close this issue.

closed 2024-10-18 by jwbonner