Mechanical-Advantage/AdvantageScope #235: Opening a log after exporting does not work

asid61, 2024-10-18:
**Describe the bug**
While the robot is running, downloading the most recent log and trying to open it via the "would you like to open this" popup does not do anything. Programs runs as if the window was just skipped.

**To Reproduce**
Download a log and click "Open Log" when the popup appears

**Expected behavior**
The log should open.

**Screenshots**

**Version (required):**
40.0.0-beta1

**Additional context**
Add any other context about the problem here.

asid61, 2024-10-18:
Version: 4.0.0-beta-1
Distributor: FRC 6328
Platform: win32-x64
Build Date: 9/26/2024, 24:41:47 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2024-10-18:
Thanks for reporting. This was fixed by 874944a4549e088ae823a9098c8dfcaa201f3b40 and will be in the next release.