Mechanical-Advantage/AdvantageScope #240: Add an option to color lines in the console tab by severity or custom RegEx pattern

katzuv, 2024-10-20:
Saving it in a custom file will be nice, so the coloring configuration will be available for all developers.

katzuv, 2024-10-21:
I'll add that it'll be nice to filter (and not only color) lines with RegEx.

closed 2024-11-02 by jwbonner

katzuv, 2024-11-04:
Thanks for the addition! I imagined something more powerful though: An ability to make a table from a RegEx pattern to a color. So for example you'll be able to color log lines by subsystem, robot mode (auto/teleop), etc. This can be saved in a JSON file that can be saved to a custom location (just like the custom assets directory), and that way it can be put in the robot code Git repository.

Here is an example from another software (LogViewPlus):
[image]
If this is too much, perhaps an option to export only the console/`DataLogManager.log()` output to a text file will be enough, then we can use another software.
Thank you!

jwbonner, 2024-11-04:
The WPILib team has discussed better methods of handling console output, with multiple channels that could be disabled or color coded separately. I think that's the best solution to this problem in the long term. Adding a button to export the console text seems reasonable though, I've opened #265 to track it.

commit 88b7456d (2024-11-02): Add option to highlight warnings and errors in console (closes #240)