Mechanical-Advantage/AdvantageScope #242: Allow adding a prefix to all NT entries when viewing a log

amsam0, 2024-10-20:
**Is your feature request related to a problem? Please describe.**
While this use case is specific to AdvantageKit, it may be useful elsewhere.

When viewing NT live, all AdvantageKit NT entries are under the `/AdvantageKit` prefix. However, when viewing logs produced by AdvantageKit, this prefix is not present. This causes all entries that have been dragged into tabs become invalid.

**Describe the solution you'd like**
It would be great to have a setting that would allow adding a prefix to all NT entries when viewing logs. For example, `/RealOutputs` would become `/AdvantageKit/RealOutputs` when viewing a log, with `AdvantageKit` being the configured prefix.

**Describe alternatives you've considered**
While there is a live source mode that only shows entries under the AdvantageKit prefix, this makes all other NT entries inaccessible. For example, you wouldn't be able to change SmartDashboard values anymore.

jwbonner, 2024-10-20:
The intended solution to this issue is the AdvantageKit NT mode. One of the issues on my backlog has been adding support for a "Tuning" table in the AdvantageKit NT mode, which could be used for editable values not managed by AdvantageKit. I've opened an [issue](243) to track that development.

closed 2024-10-20 by jwbonner

willtoth, 2025-02-16:
@jwbonner I don't quite understand how this fixes the issue being described? I also see this same behavior, that is, "RealOutputs" and all the other logging is under "NT:/AdvantageKit/" but the datalog file shows everything in the root.

This is easy to demonstrate:

1. Download the latest 2025 kitbot template code
2. Add `Logger.addDataReceiver(new WPILOGWriter());` to the SIM case in Robot.java (or run on a real robot)
3. Run simulation
4. Connect simulation to advantage scope
5. Observe data being written to NT:/AdvantageKit/ including NT:/AdvantageKit/RealOuputs
6. Close the simulation
7. Open the resulting log file saved in the `logs` directory of the project
8. Observe all data that in NT was under `NT:/AdvantageKit` is not in the root including `/RealOutputs`

jwbonner, 2025-02-16:
In the AdvantageScope preferences, try switching to the "NetworkTables 4 (AdvantageKit)" mode. It makes the `AdvantageKit` table the root, so tables like `RealOutputs` appear with the same keys as the log file. If there are values you need to edit while using that mode, AdvantageScope now supports tuning some values while using AdvantageKit ([docs](https://docs.advantagescope.org/getting-started/connect-live#tuning-with-advantagekit)).

willtoth, 2025-02-16:
I knew it would be something basic, thanks!

Small feature request. Instead of a separate setting, could you auto detect this if the key `NT:/AdvantageKit` is present?