Mechanical-Advantage/AdvantageScope #247: Visibility button state/behavior opposite what is expected

rzblue, 2024-10-22:
Currently, the show/hide button displays what the state will be after the button is pressed, so eye closed when the item is visible, and eye open when the item is hidden. This is opposite of what I expected, and of what is common in other applications, where the button indicates the current state. A few of my students that were playing with it also noticed this on their own.

A few examples:
- [Furo light mode/dark mode indicator](https://pradyunsg.me/furo)
- Photoshop/GIMP hide layer
- Google chrome star button

jwbonner, 2024-10-22:
Hmm... There are also lots of examples of toggle buttons which work the other way, such play/pause toggles and where the same eye symbols are used to toggle password visibility. There seems to be significant discussion online regarding this question, and the only universal advice is that the UI should ideally be designed in a way that is unambiguous (like using a labelled checkbox for example). I'm not sure what there's a practical solution like that given limited space, and of course the key is already grayed out for hidden items. Overall, it just seems like there's not a clear standard for what to do here?

rzblue, 2024-10-23:
I'm curious to see what others think, but for me the button indicating the action rather than the state is really disorienting and goes against muscle memory. Every time i've gone to hide/show something i've toggled it back and forth a few times because the eye makes me think it's enabled despite me seeing the data disappear.

jwbonner, 2024-10-23:
I don't feel very strongly either way, but since there seems to be some evidence that the current setup is confusing I'm open to switching it for the next beta. This button is new for beta anyway so I don't think there's any harm in hurting muscle memory for users who are used to the current approach.

The only downside is that if someone opens another issue after switching it, we'll be stuck in an infinite loop /s

closed 2024-10-28 by jwbonner