Mechanical-Advantage/AdvantageScope #255: Add the enabled/auto/teleop/etc option to the whole file

katzuv, 2024-10-23:
**Is your feature request related to a problem? Please describe.**
The log file starts before the match and ends after the match ends, but usually, only the period of the match itself (or periods within it) are interesting.

**Describe the solution you'd like**
Add the time period filter from the odometry (etc.) tabs to the whole log file.

**Describe alternatives you've considered**
While this can be done using the "locking" to auto/teleop start/end behavior of the cursor, another button will be more convenient IMO.

jwbonner, 2024-10-28:
There are filtering options for the heatmap and statistics tab because both are summarizing data without regard to timing. Every other tab is based on time-series data, with relevant options for zooming to the relevant data and playing it back. I'm not sure exactly what you're looking for regarding navigation of time-series data given the options that are already provided.

katzuv, 2024-11-04:
I was thinking about some dropdown menu that will be available for all tabs:
[image]
So if "Enabled" is selected for the line graph it will lock the earliest time to the match start timestamp and the latest time to the match end time. In the console tab, it will only show messages logged during the match, etc.

jwbonner, 2024-11-06:
It sounds like the main benefit would be quick navigation to the enabled portion of the log. I think a reasonable solution to that would be a keyboard shortcut to automatically set the visible period to the match, but retain the free navigation of the timeline that normally exists. Given that a log file can include multiple periods where a robot is enabled, a pure "filter" for enabled periods would involve making the timeline views noncontinuous, which is likely to be a confusing visual.

closed 2024-11-06 by jwbonner

katzuv, 2024-11-06:
Perhaps AdvantageScope can detect whether the log file is from a real match? I don't know if that's written to the log file currently, but teams can be instructed to log some value/print a line that indicates this is a real match (using `isFMSAttached()`).

commit f559e626 (2024-11-06): Add keyboard shortcut to zoom to enabled range (closes #255)