Mechanical-Advantage/AdvantageScope #256: Update checker bugs

katzuv, 2024-10-23:
**Describe the bug**
1. If I understand the code correctly, the update checker icon should be visible only if a newer _stable_ release is available, but it seems that's not the case: When using version `3.2.1` it suggests downloading the latest beta version, but the download button leads to `/releases/latest`, which is the latest table release.
2. In addition, when running an outdated beta release, it also leads to the latest stable release.

**Expected behavior**
1. When using a stable release, the update checker should prompt updating only if a newer stable release is available.
2. When using a beta version, lead to the latest beta release. This should also make this if statement here redundant:
  https://github.com/Mechanical-Advantage/AdvantageScope/blob/acf12ca1b2ea3eba0c4f183408fb063ce5c397af/src/main/UpdateChecker.ts#L160-L164

**Screenshots**
1.
[image]

**Version (required):**
1. `3.2.1`
2. `4.0.0-beta-1`

jwbonner, 2024-10-23:
1. This was fixed in 4.0.0-beta-1. Stable releases will only link to stable releases starting with 4.0.0.
2. This was fixed in 4.0.0-beta-2. The code you linked is the fix for this, since GitHub doesn't provide a way of linking directly to the latest pre-release.

katzuv, 2024-10-23:
2. You can just use the latest version in the API response, no? If the current version is a beta version the latest release will be a beta version, or a stable release if the current version is stable. (Now I understand the filter was added recently)

PR #258 (2024-10-28): Lead to latest appropriate release in update checker window
I'm no TypeScript expert but I think this should work.
Fixes #256.
closed 2024-10-28 by jwbonner