Mechanical-Advantage/AdvantageScope #259: Adding time sync offset key to statistics causes ascope to hang

mcm001, 2024-10-26:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

1. Load the attached wpilog
2. Create statistics page
3. Add /photonvision/.timesync/matt-desktop/offset_us or rtt2 to measurements
4. inspect devtools

**Screenshots**
[image]

Depending on time/data range:
[image]
[image]
This causes the UI to not respond to further attempts to change ranges.

**Version (required):**

Version: 4.0.0-beta-2
Distributor: FRC 6328
Platform: win32-x64
Build Date: 10/19/2024, 02:29:37 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
[ascope_kaboom.zip](https://github.com/user-attachments/files/17531679/ascope_kaboom.zip)

jwbonner, 2024-10-26:
See #238, the fix is ready for the next release.

closed 2024-10-26 by jwbonner

mcm001, 2024-10-26:
Updating to the latest beta, I can still cause ascope to hang if i change the data range to something stupid like -1e9->1000 with a step size of 100. I guess that's just a skill issue? Or how should big numbers with small variations be handled?

jwbonner, 2024-10-26:
Sounds like that’s probably a different bug that should be fixed with a sensible limit on the number of bins. Could you open another issue just to keep track?