Mechanical-Advantage/AdvantageScope #264: Support for AddressableLEDs in AdvantageScope

Strflightmight09, 2024-10-31:
It would be pretty cool to be able to see the LEDs on your robot in the simulation.

katzuv, 2024-11-04:
Not only in simulation though. I think interoperability with [Elastic's multi-color view](https://frc-elastic.gitbook.io/docs/additional-features-and-references/widgets-list-and-properties-reference#multi-color-view) will be nice. If colors can be rendered on the 3D Field tab, that'll be awesome.

catr1xLiu, 2024-11-29:
Our team also use Elastic's multi-color view to preview addressable led colors during simulation.  But it would be even better if addressable leds are displayed on the 3d robot model.

Strflightmight09, 2024-12-09:
AddressableLEDs are already visible under the `hardware` tab in the WPILib simulation, so you don't need to use Elastic's alternative