Mechanical-Advantage/AdvantageScope #266: Statistics incorrect

rzblue, 2024-11-08:
**Describe the bug**
Some statistics in the statistics tab are obviously incorrect on some topics.
Observed issues:
- min > max
- percentiles not continuously decreasing

I'm not sure if this is related or a separate issue, but it also looks like the "enabled" time range is not applying the correct filter

**To Reproduce**
Steps to reproduce the behavior:

1. Open the attached log
2. Drag `/Robot/wrist/Supply Voltage` to the stats tab
3. Select "Full log"
4. Observe Min = 10.0, max = 9.975, 1st percentile > 5th percentile, etc

Log file:
[FRC_20240316_004257_INCOL_P4.zip](https://github.com/user-attachments/files/17675777/FRC_20240316_004257_INCOL_P4.zip)

**Screenshots**
[image]
The screenshot below shows the "enabled only" time range selected:
[image]
The screenshot below shows `Robot/wrist/Supply Voltage` on a line graph
[image]

**Version (required):**
Tested on both  versions below (version included in WPILib beta 1 and latest dev build):
```
Version: 4.0.0-beta-1
Distributor: WPILib
Platform: win32-x64
Build Date: 9/26/2024, 24:43:48 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0
```
```
Version: 4.0.0-beta-2
Distributor: WPILib
Platform: win32-x64
Build Date: 11/6/2024, 21:44:18 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0
```

jwbonner, 2024-11-09:
The statistics calculations were because of [JavaScript fun](https://github.com/Mechanical-Advantage/AdvantageScope/commit/7d6b11a6cbcb89f6760fca5af7fc5c5a4b08a350) but was an easy fix. For the enabled state, AdvantageScope is using the `NT:/FMSInfo/FMSControlData` key but it doesn't seem to be updating. I'm not really sure what's happening there.

jwbonner, 2024-11-11:
The statistics issue was resolved, and the enabled data was missing from the log. As far as I can tell, that part is unrelated to AdvantageScope (but feel free to open another issue if necessary).

closed 2024-11-11 by jwbonner