Mechanical-Advantage/AdvantageScope #269: Allow overlaying a trajectory to odometry or 3d field view

spacey-sooty, 2024-11-13:
**Is your feature request related to a problem? Please describe.**
I have my Choreo trajectories in code and would like to be able to display them on the odometry view so I can view my robots tracking of them. The samples implement struct serialisable and store a pose2d but advantagescope wont use that properly to display them.

**Describe the solution you'd like**
I'd like to be able to display my trajectories using an array of samples. It would be nice to be able to display this as both a line and the robot at each point.

**Describe alternatives you've considered**
I can just log it as a Pose2d

**Additional context**
Something similar is probably possible for pathplanner

jwbonner, 2024-11-13:
PathPlanner's telemetry already logs an array of poses, which AdvantageScope can visualize. What is the format of the Choreo trajectories?

spacey-sooty, 2024-11-13:
> PathPlanner's telemetry already logs an array of poses, which AdvantageScope can visualize. What is the format of the Choreo trajectories?

Choreo trajectories are essentially a struct array with either [SwerveSamples](https://github.com/SleipnirGroup/Choreo/blob/main/choreolib/src/main/java/choreo/trajectory/SwerveSample.java) or [DifferentialSamples](https://github.com/SleipnirGroup/Choreo/blob/main/choreolib/src/main/java/choreo/trajectory/DifferentialSample.java). You can see the corresponding struct definition in each file. This would just need to pull out the Pose2d structs from the entry and use those