Mechanical-Advantage/AdvantageScope #273: Struct decoding breaks over NT with error "Fixed length arrays are unimplemented"

TheComputer314, 2024-11-15:
**Describe the bug**
[image]
The above error message appears when connecting to a sim robot over NT, breaking all struct decoding. Log files are unaffected.

**To Reproduce**
Steps to reproduce the behavior:

1. Open a robot sim with photonvision simulation.
2. Connect to the simulator.

**Expected behavior**
For struct decoding to not break.

**Version (required):**
Version: 4.0.0-beta-3
Distributor: FRC 6328
Platform: win32-x64
Build Date: 11/12/2024, 05:08:23 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
I suspect the cause is the `.photonvision` NT table?

TheComputer314, 2024-11-15:
Unsure if this is Photon's fault or AScope's fault so putting this in both repos.

jwbonner, 2024-11-16:
I think you should be able to temporarily work around this issue by avoiding publishing struct types that include arrays (specifically `OdometryFrame` seems to be the issue here).

PR #274 (2024-11-19): Warn about fixed length Photon arrays instead of throwing
Fixed length arrays are supported by non-Photon structs, which still need to be compiled by PhotonStructDecoder. This isn't an actual issue unless a Photon struct tries to use a struct with an unsupported array. Returning false just indicates that the schema cannot be compiled or used in parent Photon schemas.

Fixes #273
closed 2024-11-19 by jwbonner