Mechanical-Advantage/AdvantageScope #275: Logging a null Pose2d in C++ using structs

tssmith7, 2024-11-24:
I'm trying to remove Pose2d information for certain timestamps so things like Vision Poses don't plot with the last known logged value and "disappear" when no Pose2d data is present.

Before switching to the Struct format for complex data types I could log an empty std::span<double> to cause AdvantageScope to not draw anything (say in the odometry Tab).  Now I'm not sure how to log an "empty" or "NULL" struct for a pose in C++

Thanks.

jwbonner, 2024-11-24:
Struct logging has a distinction between logging single values and arrays of values. Numeric arrays were always treated as "arrays" of poses (so a length of 0 represented no poses). I would suggest logging your vision poses using a pose struct array, containing either 1 or 0 poses based on whether you want the values to display. This will most closely match the older behavior with numeric arrays.

tssmith7, 2024-11-24:
Got it.  Thanks

closed 2024-11-24 by tssmith7