Mechanical-Advantage/AdvantageScope #279: 3D Component Poses ignore base robot model transformations.

blaze-developer, 2024-11-25:
In our robot's model, we have to transform the robot model up for the wheels to be above the ground. However, when logging component poses, it does not respect this offset. This means we have to add an arbitrary offset to every single component pose. It would be nice to either make it respect these transformations or allow JSON configuration of these components to respect or not respect these.

jwbonner, 2024-11-25:
Component poses are always relative to the robot origin, which is defined to be on the ground. This is the standard reference frame used for everything in AdvantageScope. When following the documented setup steps, component poses can be published using that same reference frame. I'm not sure I understand what the issue is. Are you trying to use a different reference frame for publishing components?

blaze-developer, 2024-11-25:
[image]

This is my configuration. I am using a 0.045 meter z offset to get the wheels onto the ground.

[image]

My problem, is that the visualized component poses are relative to the floor, and not the offset robot pose, the components do not follow the robot with its transformations.

[image]

jwbonner, 2024-12-05:
The "actual origin" of the robot is on the floor, which is the default approach used by WPILib when converting between 2D and 3D poses (and is used by most teams). You're welcome to use a different system in your code, but AdvantageScope expects that components will be published using a standard coordinate system.

jwbonner, 2024-12-06:
Closing this issue for now, since this is the intended behavior for robot models. I added a note to the documentation clarifying the coordinate system for 3D components.

closed 2024-12-06 by jwbonner

blaze-developer, 2024-12-13:
Would you recommend making the origin of robot models in CAD on the floor in future projects to find coordinates easier without offsets?

jwbonner, 2024-12-13:
The coordinate system that you choose to use in CAD and robot code is up to you, and we recommend following the setup process in the documentation for configuring the model in AdvantageScope.