Mechanical-Advantage/AdvantageScope #281: Linking custom assets to robot

AvioNHarM, 2024-11-29:
Hey i started working on simulation using the custom assets and i followed your instructions but i run into a problem when i logged my asset position, on how do i turn it into a  component of my robot, can you pls elaborate on that?, because the docs didn't provide that much of direction.
[image]

jwbonner, 2024-12-05:
To add an object as a child of an existing object, you just drag the new field to the parent object in the list. It will be added using one of the supported child types, which you can switch by clicking on the type icon.

AvioNHarM, 2024-12-06:
oh, i get it now. thank you very much!