Mechanical-Advantage/AdvantageScope #283: Add visualization for 2d and 3d rotations

rzblue, 2024-12-04:
Currently there is no way to visualize a lone rotation. This could be useful for viewing raw IMU data.

jwbonner, 2024-12-04:
You can already view a single 2D rotation on the swerve tab, but otherwise I agree that this would be a useful addition.

ZhangzrJerry, 2025-01-12:
In fact it can in some way, I may show you our team's best practice

## Solution1: Implement a singleton class in lib

### Source code

```java
package com.nextinnovation.team8214;

import edu.wpi.first.math.geometry.Pose3d;
import edu.wpi.first.math.geometry.Transform3d;
import java.util.ArrayList;
import java.util.Comparator;
import java.util.List;
import java.util.function.Supplier;
import org.littletonrobotics.junction.Logger;

/// Class helps visualize the motion of the components in *Advantage Scope*.
/// It maintains a TF tree and stores it in a sorted list.
/// @author [ZhangzrJerry](https://github.com/zhangzrjerry)
public class AkitViz {
  /**
   * @param componentId the id of the component, ∈[0,N]
   * @param parentId the id of the parent component, ∈[-1,componentId), -1 means robot frame
   * @param transformSupplier the supplier of the transform matrix, which contains the relative pose
   *     from the parent component to the current component
   */
  public record AkitVizComponent(
      int componentId, int parentId, Supplier<Transform3d> transformSupplier) {
    public AkitVizComponent {
      // check the validity of the parameters
      if (componentId < 0) {
        throw new IllegalArgumentException("componentId out of index");
      }
      if (parentId < -1) {
        throw new IllegalArgumentException("parentId out of index");
      }
      if (isStrictBigEndian && parentId >= componentId) {
        throw new IllegalArgumentException(
            "componentId should not greater than or equal to parentId");
      }
      if (transformSupplier == null) {
        throw new IllegalArgumentException("transformSupplier should not be null");
      }
    }
  }

  /**
   * Register the component to the AkitViz
   *
   * @param akitVizComponent the component to be registered, should be declared in the constructor
   *     of the subsystem
   * @usage see {@link AkitViz}
   */
  public void register(AkitVizComponent akitVizComponent) {
    components.add(akitVizComponent);
    isSort = false;
  }

  /**
   * Should be called periodically to update the poses of the components
   *
   * @usage see {@link AkitViz}
   */
  public void periodic() {
    Pose3d[] poses = new Pose3d[components.size()];
    Transform3d[] tfs = new Transform3d[components.size()];

    if (isStrictBigEndian) {
      if (!isSort) {
[... 258 lines cut ...]
          AkitViz.getInstance().periodic();
        }
      }
    }
    ```
4. Define the zeroed transform matrix in `SubsystemConfig.java`:
    ```java
    public class SubsystemConfig {
      public static final Transform3d ZERO_POSE;
    }
    ```
5. Register a `record` object in the constructor of `Subsystem.java`:
    ```java
    public class Subsystem {
      public Subsystem() {
        AkitViz.getInstance().register(
          new AkitViz.AkitVizComponent(
            AkitViz.Ids.Subsystem.COMPONENT,
            AkitViz.Ids.Subsystem.PARENT_COMPONENT,
            () -> SubsystemConfig.ZERO_POSE.plus(this.deltaPose)  // or this.getDeltaPose()
          )
        );
      }
    }
    ```