Mechanical-Advantage/AdvantageScope #284: Deploying code freezes Ascope

asid61, 2024-12-07:
**Describe the bug**
Sometimes, deploying code causes Ascope to freeze completely. No buttons can be pressed and the readings do not update. No zooming.

**To Reproduce**
Deploy code on a non-Akit project. Sometimes it freezes (25% of the time? 50%?)

**Expected behavior**
It should never freeze irreparably.

**Screenshots**
I'll add a sc later next time it happens

**Version (required):**
Version: 4.0.0-beta-3
Distributor: FRC 6328
Platform: win32-x64
Build Date: 11/12/2024, 05:08:23 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**

jwbonner, 2024-12-13:
What tab were you using when this happened? It is occurs again, it would be helpful if you can export the layout and send it here (`File` > `Export Layout...`).

owen1050, 2025-01-17:
I have this happen almost every time I reload a simulation while the robot is enabled. Next time it happens, can I export logs somehow?

jwbonner, 2025-01-17:
What would be most helpful is to get an export of the tab layout (`File` > `Export Layout...`) plus a screenshot of what was displayed on the screen when it froze. If possible, it would also be helpful if you can send a copy of the code you were running in simulation when it happened.

blaze-developer, 2025-02-12:
This is starting to happen for my team as well, about half the time when deploying code, AdvantageScope freezes. It doesn't seem to matter which tab is open or anything of the sort, and has been happening for different tabs.

jwbonner, 2025-02-12:
Unfortunately no-one has managed to determine conditions under which this issue can be consistently recreated across multiple devices (we have spent a *significant* amount of time trying to recreate this issue).

If you can recreate the problem across multiple devices, I encourage you to upload any information you can regarding the setup (code, network conditions, *full* tab layout, etc). Until we can recreate the problem, there is very little that we can do to fix it.

blaze-developer, 2025-02-12:
Next time this happens to me, I will be sure to send as much information your way as I can. By the way, thank you for your continued support during the season, I, and many others greatly appreciate it.

Nsl106, 2025-02-17:
I am seeing this issue as well, it happens on at least the 3d field and line graph tabs. Sometimes it will freeze for a few seconds and come back as well. My vibes-based theory is that it has something to do with how long it's been connected for, I think it happens less with quick redeploys vs longer connections. I'm currently testing in simulation, but it has happened on a real robot as well. This was a problem for us with several non-standard code setups in the 2024 offseason (which I originally blamed the problem on), so I'm pretty sure it's not codebase dependent.

Here is a screenshot of the frozen screen + the current tab layout:
[AdvantageScope 2-16-2025.json](https://github.com/user-attachments/files/18816946/AdvantageScope.2-16-2025.json)

[image]

blaze-developer, 2025-02-24:
To add, this seems to affect AdvantageScope instances on devices that are separate from the device that is initiating the deploy sequence. Every device connected to the network through AdvantageScope freezes on deploys. I do not have any layout JSON but we've only really had line-graph and 3d fields open when this has happened. If I get the chance, I will probably try deploying with different tabs open to see if the current tab makes a difference.

Tuna156, 2025-02-26:
Some more info if it helps. It seems that the idle state of AdvantageScope goes on for a very long time when AdvantageScope freeze? I can't really comprehend the code in here that well. When it's in this idle state it runs code in NT4Source.ts but I can't find any code in AdvantageScope that makes the code go into idle itself

[image]

Tuna156, 2025-02-28:
Sorry for the delay I have been very caught up in school work but I think I found the issue. LogField::clearBeforeTime() spends an enormous amount of time shifting new arrays in this while loop. The freezing duration seems inconsistent, probably has to do with how much data AdvantageScope is recording. The last time it was a whole minute but this time it was only about 8 seconds

[image]

Tuna156, 2025-03-01:
I made a PR that should in theory fix this issue. I have tested this patch for a few hours and I haven't run into this deploying problem anymore so it should be fine but please test it out if you please

PR #336 (2025-03-08): Stop AdvantageScope from occasionally freezing on deploy
* Closes #284 where deploying/simulating robot code causes AdvantageScope to freeze occasionally.
* Apologies for yet another small commit again, but this PR optimizes LogField::clearBeforeTime so that it is not re-indexing the 2 arrays every loop inside the while loop.
* I've been testing this for a few hours, and I am no longer able to recreate the original issue, however, if others could test it out to be sure that would be very appreciated!
closed 2025-03-08 by jwbonner