Mechanical-Advantage/AdvantageScope #287: Playback Speed Slider

alexanderszura, 2024-12-11:
We used this data logging software yesterday to review drive team tryouts where we had the thought of a speed slider, speeding up the replay or slowing down the replay.

If a slider could not be achieved for some reason then maybe a "speed selector" where it has a preset number of values (eg. 1x, 2x 0.5x)

jwbonner, 2024-12-11:
AdvantageScope allows the playback speed to be changed by right-clicking the play/pause icon in the title bar. It uses the approach of a set of presets, largely because this allows for more logarithmic options (e.g. 200%, 400%, and 800% are all options, but very few users are likely to want 500% speed).

alexanderszura, 2024-12-11:
Ahh well that's aweoms! Thank you so much!

closed 2024-12-11 by alexanderszura