Mechanical-Advantage/AdvantageScope #303: Rotation Arrow Missing in Swerve View When Translational Speeds Are Zero When Displaying Chassis Speeds

MichaelLesirge, 2025-01-07:
**Describe the bug**

In the Swerve View tab of the 2025 Advantage Scope (WPILib 2025), the rotation arrow for a chassis speeds object only appears when the vx and vy components are large enough to draw a translation arrow. Specifically, the rotation arrow is missing when there is significant rotational speed but negligible translational speed.

**To Reproduce**

1. Open the Swerve View tab (shortcut: Alt+D).
2. Log a chassis speeds object with a high omegaRadiansPerSecond value but very small vxMetersPerSecond and vyMetersPerSecond values. Example: `Logger.recordOutput("SwerveStates/DesiredRobotSpeeds", new ChassisSpeeds(0, 0, 5));`
3. Add chassis speeds entry (`SwerveStates/DesiredRobotSpeeds`) to the sources in the Swerve View tab.
4. Observe that the rotation arrow is not displayed.

**Expected behavior**
The rotation arrow should appear even when there is no translational speed.

**Screenshots**

Screenshot showing the bug:
Despite a rotational speed being present in chassis speeds, the rotation arrow is missing.[image]

Screenshot showing correct behavior, showing case which works:
The rotation arrow appears correctly when there is both rotational and translational speed, meaning it works most of the time. I added this screenshot to the bug report just to show what the working case is like, to compare to not working case.
[image]

**Version (required)**

Version: 4.1.0
Distributor: WPILib
Platform: win32-x64
Build Date: 12/31/2024, 18:23:20 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**

This behavior was a bit confusing at first, but it’s not a major issue. I wanted to report it to help improve this fantastic feature. I really appreciate the work you’ve put into Advantage Scope—it’s been incredibly useful!

MichaelLesirge, 2025-01-07:
Hopefully I am not misinterpreting intended behavior, but if this is intended it would be nice to have a way to turn it off.

jwbonner, 2025-01-07:
Thanks for reporting that, it's definitely not the intended behavior. I just pushed a fix which will be part of the next release.

closed 2025-01-07 by jwbonner