Mechanical-Advantage/AdvantageScope #311: AdvantageScope XR - Rescale the Field Projection for Half-Court Spaces

LTCjRet, 2025-01-14:
Regarding projections of 1:1 scale fields, with the AdvatageScope XR app on iPhone/iPads it would be helpful if the drag-and-drop sizing of the field was done from the drivers’ station wall to mid-court instead of for the entire field (Alliance wall to Alliance wall).

I suspect the vast majority of teams are space limited. For those that wish to use 1:1 scale AR projections for a half field, it is impossible to scale it in an area only sufficient for a half field since the user can’t move to the position of the opposite alliance wall. Such a method would not limit a 1:1 full scale field and it would open up use to many other teams. Additionally, this year’s game is largely a half-field game anyway.

jwbonner, 2025-01-14:
When using the full-size calibration mode, you measure two points along an alliance wall plus a point anywhere along one of the two long field barriers. This only requires one alliance wall and one field barrier (a single corner of the field). It seems like this addresses your concern if I’m understanding correctly.

jwbonner, 2025-01-17:
Closing because I think this is addressed by the current implementation.

closed 2025-01-17 by jwbonner