Mechanical-Advantage/AdvantageScope #312: 3D Field MechanismLigament2D rendering breaks with length 0

mjansen4857, 2025-01-15:
**Describe the bug**
When displaying a Mechanism2D in the 3D field tab, if a ligament has length 0, it will no longer be rendered even if the length is later set to a non-zero value. Reloading the robot model by changing its model/type will allow it to render again, but it will break if that ligament ever has a zero length.

**To Reproduce**
Steps to reproduce the behavior:

1. Create a Mechanism2D with a single ligament with an initial non-zero length
2. View the mechanism on the 3D field tab, notice the mechanism is rendered
3. Set the ligament length to 0
4. Set the ligament length to a non-zero value again
5. Ligament is no longer visible

**Version (required):**
Version: 4.1.1
Distributor: FRC 6328
Platform: darwin-arm64
Build Date: 1/9/2025, 20:08:20 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

cyocom, 2025-01-15:
I was also able to reproduce this

Version: 4.1.1
Distributor: FRC 6328
Platform: win32-x64
Build Date: 1/9/2025, 20:08:20 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2025-01-15:
Thanks for catching that. I pushed a fix, which will be in the next release.

closed 2025-01-15 by jwbonner