Mechanical-Advantage/AdvantageScope #313: Cant get tuning working when in NetworkTables 4 (AdvantageKit)

NateDCoder, 2025-01-15:
I am using the TalonFX AdvantageKit template project and when I copy the code from https://docs.advantagescope.org/getting-started/connect-live/#tuning-with-advantagekit it just creates a folder inside NetworkInputs the slider icon also disappears when switching form 
NetworkTables 4 and NetworkTables 4 (AdvantageKit). I am also changing the Network type by click on Help>Show Preferences and wondering if I am just doing somethings wrong.

jwbonner, 2025-01-15:
The `LoggedNetworkNumber` class doesn't actually publish a value to NetworkTables until a default value is set, either through the constructor or by calling `setDefault`. I've updated the docs to reflect that. Try doing this instead and see if that fixes the problem:

```java
LoggedNetworkNumber tunableNumber = new LoggedNetworkNumber("/Tuning/MyTunableNumber", 0.0);
```

NateDCoder, 2025-01-15:
Thanks for the quick response and yes I tested it, and it did work, though it only worked under NetworkTables 4 and not NetworkTables 4 (AdvantageKit)

NateDCoder, 2025-01-15:
Thank for the quick response that fixed the issue

closed 2025-01-15 by NateDCoder