Mechanical-Advantage/AdvantageScope #314: Why does AdvantageScope require Emscripten to compile?

ancientstraits, 2025-01-16:
I was surprised when I saw in the `README.md` that AdvantageScope requires Emscripten to compile. I checked the git repo to see what part of it requires Emscripten. Please correct me if I am wrong, but I think that the only thing that Emscripten is used for in AdvantageScope is the `wpilogIndexer.c` file. Are there any other places where wasm is used? If not, then why isn't the `wpilogIndexer.c` file written in TypeScript instead of C? I am guessing it is for performance, but I do not know if TypeScript is *that* much slower than C is. If you need the C file to be rewritten in TypeScript, I can help, since I know both C and TypeScript well.

jwbonner, 2025-01-16:
Yes, the WPILOG indexer is the only WebAssembly piece of AdvantageScope. The performance difference between TypeScript and WebAssembly is very significant in this case: it speeds up log parsing by roughly 2x compared to the old TypeScript implementation (which had already been through multiple rounds of optimization over several years).

closed 2025-01-16 by jwbonner

ancientstraits, 2025-01-16:
If there are any other parts of AdvantageScope that would probably be faster if rewritten in C, please let me know.