Mechanical-Advantage/AdvantageScope #317: Mechanism2D bug in 3D Field if the length is set to 0

MrTinker64, 2025-01-21:
**Describe the bug**
If at any time the Mechanism2D is set to a length of 0 and then set to a non-zero length it disappears from the 3D field view, but not the 2D mechanism view.

**To Reproduce**
Steps to reproduce the behavior:

1. Create a Mechanism2D, a root, and a ligament with a non-zero starting height
2. In robotPeriodic log the mechanism as an output and set the ligaments length to 0 when one button is pressed and any non-zero number when that button is released
3. Attach the Mechanism2D to a robot in AdvantageScope's 3D field view
4. Press the button and release it, the mechanism will go to the set length then go to zero (as expected). Then try pressing it again, this time the mechanism will no longer be visible (unexpected).

**Expected behavior**
The Mechanism should go to zero length when I set it to zero and then the non-zero length I set it to after that.

**Screenshots**
Example of Mechanism2D view working while the 3D view is broken
[image]

Example of the 3D view while Mechanism is set to 0. Note that the angle of the mechanism is no longer working either.
[image]

**Version (required):**
Version: 4.1.0
Distributor: WPILib
Platform: darwin-arm64
Build Date: 12/31/2024, 18:23:20 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
This also happens if the Mechanism has an initial length of zero instead of being set to 0 after having a greater initial value.

jwbonner, 2025-01-21:
Duplicate of #312, which is fixed and will be in the next release. You can download the dev builds [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/workflows/build.yml?query=branch%3Amain) if you want to use the fixed version in the meantime.

closed 2025-01-21 by jwbonner

MrTinker64, 2025-01-21:
dang it I forgot to search closed issues 🤦, sorry!