Mechanical-Advantage/AdvantageScope #318: Adding second Mech2d to robot changes positioning of first

mjansen4857, 2025-01-21:
**Describe the bug**
Adding a second Mechanism2d to a robot model on the 3d field can affect the positioning of other mechanisms already added to the robot. Basically, the origin is being shifted for some reason. The mechanisms have different canvas sizes, which may be part of the issue. My guess is that the positioning of the mechanism on the smaller canvas is being mapped on to the larger canvas, which is no longer centered.

**To Reproduce**
Steps to reproduce the behavior:

1. Create a mechanism 2d and add it to the robot. For us, this first mechanism has a canvas size of (0.5, 2.25)
2. Notice mechanism is centered on robot model
3. Create a second mechanism 2d and add it to the robot. For us, this second mechanism has a canvas size of (1.0, 1.0)
4. Notice first mechanism has been shifted off center

**Screenshots**
One mechanism:
[image]

Two mechanisms:
[image]

**Version (required):**
Version: 4.1.1
Distributor: FRC 6328
Platform: darwin-arm64
Build Date: 1/9/2025, 20:08:20 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

mjansen4857, 2025-01-21:
Can confirm that making the X dimensions of both canvases match prevents this issue.

jwbonner, 2025-01-24:
Thanks for catching that. I switched the logic when merging mechanism states to automatically align using the center of the bottom edge rather than the lower left, which should address this issue.

closed 2025-01-24 by jwbonner