Mechanical-Advantage/AdvantageScope #325: Units Library Logged Values

blaze-developer, 2025-02-06:
Hi! Not sure if this is a AdvantageScope, or AdvantageKit request, or both.

When logging information in unit library measures, my understanding is that right now, AdvantageKit logs the base unit. This can be undesirable because when looking at a log file, you may not know what base unit a certain field is using. It would be great if when these values were logged, they passed down some information about their base unit, and AdvantageScope could use that data to show the user what the base unit of a field is.

This, along with #121 would make unit conversions and log review much more streamlined.

This could be accomplished by having a custom serialized class for unit measures, instead of simply logging the base unit.

Kevok93, 2025-02-07:
+1

To my understanding of the intent of the Units/Measures library, the base unit is intended to be undecipherable, with the recipient of a Measure value being the one who needs to explicitly decode the Measure object back into whatever units the recipient wants, therefore I would advocate for AdvantageScope being able to choose which units to decode the Measure object into.

reecelikesramen, 2025-03-23:
+1 for sending the unit over NT

closed 2025-09-18 by blaze-developer