Mechanical-Advantage/AdvantageScope #326: Option to allow repeats in AdvantageScope Console Tab

blaze-developer, 2025-02-09:
I am console-logging data in a simulation to verify that triggers are being triggered at the correct times. However, it only shows up once in AdvantageScope. I suspect that it filters out duplicates. This is nice, however sometimes duplicates are necessary within the console to see when things are being logged.

[image]

[image]

blaze-developer, 2025-02-17:
Alternatively, append to the previous line a multiplier. (like for example if it logged the same line 4 times, say x4.

derekriter, 2025-06-08:
I second this. This could definitely cause problems when reviewing console logs after a match.