Mechanical-Advantage/AdvantageScope #327: Download Logs menu shows non-existent files

mjansen4857, 2025-02-10:
**Describe the bug**
The download logs menu in AdvantageScope is showing all of our logs from last year, both wpilog and hoot files. The USB drive has been wiped and formatted, and the RIO SD card has since been flashed. These files are not present anywhere on the robot. So, it seems that AdvantageScope might be caching the list of files and displaying them even if they are no longer present. Trying to delete these non-existent files from the download logs menu does nothing, and they will still be shown in the list.

**To Reproduce**
Steps to reproduce the behavior:

1. Open download logs menu when connected to robot with logs present on USB drive
2. Log files on drive are shown in list
3. Manually delete log files from USB drive without using the download logs menu
4. With USB plugged in to RIO again, connect and open download logs menu
5. Deleted log files are still displayed in list

**Expected behavior**
Only log files currently on the RIO's sd card or USB drive should be shown

**Version (required):**
I don't have access to the laptop we saw this issue on at the moment so I can't get the version output. But, we are using the latest release on windows x86. I should be able to get this info tomorrow. Not sure if this issue is present on other platforms/systems.

jwbonner, 2025-02-11:
That's a very strange issue. AdvantageScope doesn't cache the list of files in the download window, and as far as I know the underlying library ([ssh2](https://www.npmjs.com/package/ssh2)) doesn't cache that data either. In my testing, I'm able to see files disappear in the downloads window when they are removed from the directory.

Are you able to see the files when connecting using a different SFTP client? Have you confirmed that you are definitely connected to the correct roboRIO (and not, for example, another RIO with the `.2` IP address)?

mjansen4857, 2025-02-11:
I'll give another SFTP client a try tonight. I do know at least that CTRE's Tuner X does not show any of the old hoot files shown in AdvantageScope from their log downloader tool. Looking at the files by plugging the drive into a laptop does not show old logs either. Definitely connected to the right RIO.

mjansen4857, 2025-02-12:
So we found out this was due to some weird partition shenanigans somehow? No idea how, but the drive ended up with 2 partitions at some point. The RIO would mount one partition, which had all the old log files. Windows would mount a different one, which was empty. Weirdly, the partition containing logs never showed up in windows and disk manager showed the entire drive was only the one empty partition, so formatting the drive didn't even remove the "real" partition with logs.

We were able to reformat the whole drive with a different application that got rid of the weird partition stuff and old files.

closed 2025-02-12 by mjansen4857