Mechanical-Advantage/AdvantageScope #335: AdvantageScope exported wpilog files lossy, not useful for SysId

bhall-ctre, 2025-02-27:
AdvantageScope currently has two options for exporting to wpilog:
1. Only changes in signal values
2. Values sampled at a fixed period.

However, SysId expects all signals to be continuously updated (even if unchanged), and sampling at a fixed period is lossy; you lose some data and the original timestamps. Some of our users on non-Windows platforms are using AdvantageScope to convert hoot files to wpilog, rather than using our `owlet` tool directly, which causes issues in SysId. We intend to update our documentation to make the usage of owlet more clear and recommend using that to properly export to wpilog.

However, it may also be useful for AdvantageScope to add a "lossless" export option that re-exports all signal values with their original timestamps, regardless of if the values changed. For example, that could be useful if combining multiple wpilogs together. Otherwise, AdvantageScope documentation should be updated to warn against using AScope-exported wpilogs in SysId.

bhall-ctre, 2025-02-27:
> Can you elaborate on the "issues" caused when using SysId?

SysId currently uses the `voltage` field as the basis for timestamps and linear interpolation. As a result, if the motor voltage manages to be constant across 2+ log writes, then all the position/velocity data between those log writes will be lost using "All Changes". Of course, this is a much bigger issue in simulation (which as you pointed out isn't an intended use case of SysId), where SysId ends up [only seeing a few samples](https://www.chiefdelphi.com/t/hoot-log-for-sysid-is-missing-data/493424) and fails. However, it is still technically possible with hardware on a short timescale, even if highly improbable. And simulation with SysId can still be a useful exercise in learning the tool.

jwbonner, 2025-02-27:
I'm fine with updating the AdvantageScope documentation to recommend not using it for exporting simulated Hoot logs specifically (AdvantageKit already solves this problem). I can also mention the potential for data loss, though with the relatively low likelihood of that occurring I also want to avoid confusing users about the actual scale of potential problems. We can probably implement the proper fix for 2026, but until then I don't think it's in the best interest of our users to broadly recommend the *much* more inconvenient path of using the `owlet` CLI (since we also can't assume that our users have access to Tuner X on the same device).

jwbonner, 2025-03-08:
I added a section to the documentation providing more information about this issue, which will be published as part of the next AdvantageScope update.

closed 2025-03-08 by jwbonner