Mechanical-Advantage/AdvantageScope #339: wpilog and .hoot file timestamp alignment

jashow55, 2025-03-03:
**Describe the bug**
We record with advantage kit and want to align to .hoot file from CANivore bus, but the two files are not aligning when opened.

**To Reproduce**
record during a match with your project with AdvantageKit and signallogger , 
copied the .wpilog and .hoot from the USB stick to PC.
open together in AdvantageScope

**Expected behavior**
The autonomous and telelop from both files align.  
Right now the recorded file from akit starts when the robot is turned on.
The signal logger starts, it looks like, when the match starts.

[image]

**Version (required):**
AdvantageScope 4.1.2, Windows 11
AdvantageKit 4.1.1
CTRE firmware 25.3.0.0

**Additional context**
AdvantageScope documentation says the files should auto-align based on timestamp.    
I can provide the two files I am trying to sync.

jwbonner, 2025-03-04:
Thanks for reporting. It would be helpful if you can send the two log files that trigger the issue. You may need to put them in a zip file to upload to GitHub.

jashow55, 2025-03-04:
[logs.zip](https://github.com/user-attachments/files/19062838/logs.zip)

I included the rio .hoot file too.   But it is the CANivore drivebus .hoot and the .wpilog from akit that is in the picture.

jashow55, 2025-03-04:
if it helps, our source code is at:
[https://github.com/FRCTeam68/2025-CompBot](url)

we based off of advantagekit swerve template this year.

We are not enabling signallogger but I understand it is automatically enabled in matches if you don't intentionally disable it.

jwbonner, 2025-03-08:
Thanks for catching that. I pushed a fix, which will be in the next release very soon.

closed 2025-03-08 by jwbonner