Mechanical-Advantage/AdvantageScope #340: Scrolling through table render issues

rjbell4, 2025-03-05:
**Describe the bug**
When scrolling around a table, sometimes the display "freaks out", and rapidly alternates between neighboring scroll positions, without any keyboard or mouse input.  A video will be provided below.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to a table view in AdvantageScope
2. Scroll a lot, seemingly mostly using arrow keys a lot
3. See error
4. Sometimes, scrolling more with mouse can get the problem to terminate, through it often comes back with limited additional scrolling

**Expected behavior**
Screen continues to be rendering normally.

**Screenshots**
https://github.com/user-attachments/assets/ea74db3b-401f-4173-835c-0eb079b61a8a

**Version (required):**
Version: 4.1.2
Distributor: FRC 6328
Platform: win32-x64
Build Date: 2/1/2025, 24:04:26 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

rjbell4, 2025-03-05:
I was able to reproduce the issue and record a video with my phone, which is a somewhat clearer depiction of the issue, though it still suffers from different refresh rates.

https://photos.app.goo.gl/UYGpP2E1Ev81XDQU7

jwbonner, 2025-03-08:
I fixed an issue where boolean values would cause the row heights to change and trigger some visual problems. I *think* that should fix this issue with the view flipping back and forth (it hasn't happened since I made the fix), but I haven't been able to recreate it consistently so I'm not certain. Feel free to reopen if this reappears.

closed 2025-03-08 by jwbonner